recalbox/mk_arcade_joystick_rpi #2: Install problems

mal3kith-developer, 2015-01-22:
Hi 
First:
i try to install this driver behind berryboot and retropie default installation, after sudo sh ./install.sh command ill get errors, log says about unrecognized  kernel: 3.12.26-rpi-aufs, which is default installed with berryboot.
Second:
Alone retropie img install (kernel 3.12.22+) dont works with your driver.

Can you help me please?

Thanks in advance

digitalLumberjack, 2015-01-22:
I'll look at it tomorrow.

digitalLumberjack, 2015-01-24:
Hi,

The issue come from the kernel in retropie image is 3.18.X and no headers armhf are available. You need to install them by hand for now, i'll make some modifications in the installation file to be more specific on the error, and i'll try to publish headers when i'll have some time.

digitalLumberjack, 2015-01-27:
Ok so we saw on the other issue (https://github.com/digitalLumberjack/mk_arcade_joystick_rpi/issues/3) that we cannot install the module on 3.18.X kernels.

If you are on raspbian, you have to get the 3.12.28+ kernel by downgrading your rpi-firmware with : 
```sudo rpi-update de69b134dc6e4066fe70db29816d57895dffd9b9 ```
Then reboot and use ```uname -r ``` to check the version of you rpi.

No you can compile with ./install.sh

For other os i'm trying to find a way to make the diver compatible.

mal3kith-developer, 2015-01-27:
I'm using retropie img, i dont know if it is raspbian based so. But i'll
try anyway.

Thank you

digitalLumberjack, 2015-01-28:
The last retropie beta should now work as it has the 3.18.3 kernel.

closed 2015-01-28 by digitalLumberjack

mal3kith-developer, 2015-01-28:
Thank you very much

mal3kith-developer, 2015-01-30:
Hi, ill install it yesterday and works fine.

Thank you again

2015-01-28 9:10 GMT+01:00 Male Kith <mal3kith.developer@gmail.com>:

digitalLumberjack, 2015-01-30:
Great news :)