recalbox/mk_arcade_joystick_rpi #3: Can't insert the module

musenwoody, 2015-01-25:
Hi.

For there is no headers armhf for 3.12.36+, I used 3.12.35+:
[image]

I was able to install it without problems, but it couldn't be inserted:
[image]

and in dmesg:
[image]

I found in modinfo:
[image]

The vermagic was 3.12.35 but not 3.12.35+. 
At this point I am assuming it is because the module I got is 3.12.35 but my Rpi is running 3.12.35+?

I tried modprobe --force-vermagic and got the same error:
[image]

I tried the alternative option and get another error:
[image]

This is what in make.log:
[image]

Would anyone please point me in the right direction?

digitalLumberjack, 2015-01-26:
Hi,

Ok i modified the make few days ago. This is what causes the compîlation error. I'm going to make so changes in paralel to the other bug.

I'll let you know

musenwoody, 2015-01-26:
Hi.

Thank you for your reply.I will try to compile again when you update the make.

And I want to know that if the vermagic was wrong when I tried to insert module.
How could I get the right vermagic matched my kernel?

Can you help me please?

Thanks in advance

digitalLumberjack, 2015-01-26:
I think the only way to recompile the module without having headers available in apt, is to recompile the whole kernel, to have a Module.symvers created, and then recompile the module.
Ugly.

You are 3.12.35+ check at http://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/ you headers are available.

musenwoody, 2015-01-26:
Thank you for your reply.

I do have 3.12.35+ headers installed.
[image]

But After I installed the driver I still couldn't insert it because of the wrong vermagic in modinfo:
3.12.35

Is there something wrong with what I have done?

digitalLumberjack, 2015-01-26:
Hi pal, i restore the first makefile, could you try to reclone the repository and try to make the module ?

musenwoody, 2015-01-26:
Thank you!

I'll have a try right now

musenwoody, 2015-01-26:
It seems that the Makefile works:
[image]

but...
[image]

and in dmesg:
[image]

modinfo:
[image]

musenwoody, 2015-01-26:
Hi.

I think the problem is there is something wrong with the module i compiled.Because I only get the vermagic "3.12.35".

And vermagic of the other modules worked on the Rpi is "3.12.35+"
For examples:
[image]

Any opinions?

digitalLumberjack, 2015-01-27:
I changed my kernel to 3.12.35+, installed corresponding headers, tried to build the module. I have the same error as you.

And i can't explain yet :(

musenwoody, 2015-01-27:
Thank you for your help. Maybe I should try another kernel . 

Which version of kernel do you suggest to compile the module?

BTW, I downloaded the new RetroPie image(ver :2.4.2 betw) today. And I found that the kernel is 3.12.35+ too ...

digitalLumberjack, 2015-01-27:
i'm goning to try the 3.12.2X or something like that. This issue here is that i wrote the driver with my c c++ skills but when it comes to kernel revision version errors with the good kernel and the good headers, i'm not so good...

digitalLumberjack, 2015-01-27:
I'll try to install db9 or gamecon from http://www.niksula.hut.fi/~mhiienka/Rpi/ . I based the installation script and package on thoses. If they work i'll be able to correct the error.

musenwoody, 2015-01-27:
Hi pal, you have done a great job of course !

I hope I could help with your project.I'll let you know once I make further progress.

digitalLumberjack, 2015-01-27:
Ok, it works really great with kernel and headers 3.12.28+

digitalLumberjack, 2015-01-27:
I'm contacting marqs, the original author of db9 driver, he publishes the headers on http://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/ too.

Maybe he can explain.

digitalLumberjack, 2015-01-27:
Just for info, by configuring the Makefile in /lib/modules/3.12.38+/build/Makefile to have the + beside the version, the vermagic is ok, but that doesn't change nothing, the module won't load.

musenwoody, 2015-01-27:
Thank you for your reply!

I will try 3.12.28+ , and 3.12.38+ later.

digitalLumberjack, 2015-01-27:
Tell me if it's ok with 3.12.28+ ;)

The second solution would be to compile your own kernel and install it. But on RPi, it would take about 12 hours.

musenwoody, 2015-01-27:
I will reply ASAP :)

musenwoody, 2015-01-27:
Hi pal.

I got the same error on 3.12.28+

digitalLumberjack, 2015-01-28:
I released a 0.1.3 version.

marqs updated it's header generator script, where the Module.symvers fetched was not the good one.

Now it works with 3.18.X kernels

closed 2015-01-28 by digitalLumberjack

digitalLumberjack, 2015-01-28:
I'll be glad if you can test and confirm that :)

musenwoody, 2015-01-28:
Hi. Thank you for your update!
I will test it wiith pleasure!

musenwoody, 2015-01-28:
Finally it worked on 3.18.3+.

Thank you for your great work!

digitalLumberjack, 2015-01-28:
Super good !