recalbox/mk_arcade_joystick_rpi #4: Not working for RPi2 - 3.18.7-v7+

Jesserockz, 2015-02-13:
Hi, I'm sorta new to linux so not sure if its just me.

I have a RPi2 here with RetroPie working. I can install the module if my SD is in a RPi1 B but when I put it into the RPi2 it will install but not load.

pi@raspberrypi ~/mk_arcade_joystick_rpi $ sudo modprobe mk_arcade_joystick_rpi map=0x20
ERROR: could not insert 'mk_arcade_joystick_rpi': Exec format error

pi@raspberrypi ~/mk_arcade_joystick_rpi $ modinfo mk_arcade_joystick_rpi
filename:       /lib/modules/3.18.7-v7+/updates/dkms/mk_arcade_joystick_rpi.ko
license:        GPL
description:    GPIO and MCP23017 Arcade Joystick Driver
author:         Matthieu Proucelle
srcversion:     2B2E40AFD619744883AA3C0
depends:
vermagic:       3.18.7 preempt mod_unload modversions ARMv6
parm:           map:Enable or disable GPIO and MCP23017 Arcade Joystick (array of int)

dmesg output: [ 4037.517855] mk_arcade_joystick_rpi: disagrees about version of symbol module_layout

I have tried using the install script - which fails because there is no header file for 3.18.7-v7+ 
I have downloaded the 3.18.7+ headers and linked -v7+ to that (not sure if that is completely wrong). Also have tried downloading raspberrypi/linux with no luck. 
I have also tried cloning the repo and running makepackage.sh and the dpkg -i, but I am not having any luck.

Thanks for the great work so far.

digitalLumberjack, 2015-02-13:
Yes you will have to recompile the kernel and install it if you want to have the current headers. 
RPI2 headers are not available for now.
Maybe marqs will ditribute armv7 headers, until then recompile the kernel : 
http://elinux.org/Raspberry_Pi_Kernel_Compilation

Jesserockz, 2015-02-13:
Thank you, I see that they have now been uploaded: http://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi2/linux-headers-3.18.7-v7%2b_3.18.7-v7%2b-2_armhf.deb
I will test this and get back.

Jesserockz, 2015-02-13:
Hi again, I have just tried using the new headers, but it still seems modprobe will not load it. I did modify the script to download the rpi2 headers.

pi@raspberrypi ~/mkjoystick $ sudo modprobe mk_arcade_joystick_rpi map=0x20
ERROR: could not insert 'mk_arcade_joystick_rpi': Exec format error

dmesg: [ 2378.181081] mk_arcade_joystick_rpi: disagrees about version of symbol module_layout

pi@raspberrypi ~/mkjoystick $ modinfo mk_arcade_joystick_rpi
filename:       /lib/modules/3.18.7-v7+/updates/dkms/mk_arcade_joystick_rpi.ko
license:        GPL
description:    GPIO and MCP23017 Arcade Joystick Driver
author:         Matthieu Proucelle
srcversion:     2B2E40AFD619744883AA3C0
depends:
vermagic:       3.18.7-v7 SMP preempt mod_unload modversions ARMv7
parm:           map:Enable or disable GPIO and MCP23017 Arcade Joystick (array of int)

digitalLumberjack, 2015-02-14:
Ok i'm checking with marqs and i'll be back soon.

digitalLumberjack, 2015-02-14:
And it's ok, new kernel headers compile well and the drivers too :) 
Thanks to marqs ;)

Jesserockz, 2015-02-14:
Yay, its working. Thank you and marqs very much. Now I want to try to add more 'buttons' to each joystick. Like a credit button for a coin mech. And want to get it working in kodi.

It will be in an old arcade cabinet at some point. At the moment its just the bare RPi and a breadboard made gamepad for testing. So photos yet, but I may make a post later on somewhere.

closed 2015-02-14 by Jesserockz

digitalLumberjack, 2015-02-15:
You will see that i updated the installation procedure, it's now the same for RPi1 and RPi2.

You already have 8 button per player, what is do is using select as COIN for all arcade games. As there are no arcade cabinet with more than 6 game button per player and 1 for coin and 1 for start, i though 8 buttons per player would be suffisent.

I'm courious why you want more ?

Jesserockz, 2015-02-15:
Yes I have seen that, Thank you. I actually got a photo today of the cabinet it will be going into and each player has 6 buttons plus start and a single coin mech.
This is the first time I have attempted anything like this. Now that I know that, I guess it doesn't need more.
I was also hoping I could use this in kodi/xbmc but looks like getting gamepad input is not viable for RPi.

digitalLumberjack, 2015-02-15:
It's a mess with XBMC, but it's possible even without joystick support. You will have to create an event client and load the mk_arcade_joystick_rpi AFTER xbmc.
See and example here https://github.com/xbmc/xbmc-rbp/blob/master/tools/EventClients/Clients/XBMC%20Send/xbmc-send.py

Jesserockz, 2015-02-15:
Oh ok, cool thanks for that.