recalbox/mk_arcade_joystick_rpi #5: Rpi B - "left" button not working

drygol, 2015-02-20:
Hi

I recently downloaded you driver. I compiled it successfully and implemented it. How ever I am having a problem with "LEFT" button not working. I already eliminated hardware - cables and button itself.
I even recompiled the driver with different GPIO assigned and unfortunately no luck there as well :(
JSTEST works fine for all buttons except "LEFT" - GPIO27 by default.
Do you have any suggestion how to debug this issue ? 
Maybe something changed in code with your latest submission and support to RPI2 ? 

Cheers
Pete

drygol, 2015-02-21:
I also posted similar question on rpi forums but nobody seems to have this issue.

http://93.93.128.176/forums/viewtopic.php?f=78&t=82878&sid=d0d74e743d877f9f3d166b283daee1d2&start=50

I wonder if my RPI B is bricked or is it simply a bug after you recent commits,

digitalLumberjack, 2015-02-21:
Your pi may be bricked as it's running on many pi for me, i installed it on a b+ yesterday...

digitalLumberjack, 2015-02-21:
Try that : unload the driver, 
then run with python : 
````
import RPi.GPIO as GPIO

GPIO.setmode(GPIO.BCM)
GPIO.setup(27, GPIO.IN, pull_up_down = GPIO.PUD_UP)
GPIO.wait_for_edge(27, GPIO.FALLING)
print("GPIO 27 is ok")
GPIO.cleanup()
```

drygol, 2015-02-21:
Hmm , I should have figure that out earlier :D Thanks for a hint to check GPIOs first lol 
I just gave it a go and python script never ends , it just runs forever. I am not familiar with RPi gpio python libraries and from a code I can see that there is no error handling except success. If script runs forever , does it mean that GPIO 27 is bricked ? I already tried with other GPIOs and a result is pretty much the same. Also I just installed wiringPI and it shows that all user GPIOs are OK but it skips 27 so I will have to look for some other test or simply recode stuff to fit my check. Could you please give me a hint if above code returns an error in case of faulty GPIO ?

drygol, 2015-02-22:
Ok m I just figured that python code out.  So it looks like it is not working if I set GPIO 27 - os reports that this GPIO is already assigned and busy.However if I put there 21 instead it all works correctly. GPIO 21 is a same pin as 27. I just changed  mapping to 21 instaed 27 for "left " in mk_arcade_joystick_rpi.c and built a deb package and installed it. Unfortunately it is still not working.
So RPi is not bricked.

digitalLumberjack, 2015-02-25:
Ok. If you test the python code with gpio22 (right) does it work ? (don't forget to unload the driver)
That's weird it looks like something is using gpio 27 for a special feature...

drygol, 2015-02-26:
Yup , I did unload that module prior to testing.
All other standard/user gpio`s work flawlessly except that 27 one. I`ll install fresh retropie or raspbian and check it again `cos it starts to looks like that something is eating up this gpio  like you just said.
I`ll report back after test ;)

digitalLumberjack, 2015-03-20:
Any progressions @drygol ?

vitorveras, 2015-05-17:
Same problem here on RPI2. jstest shows :Axes:  0:-32767  1:     0 Buttons:  0:off  1:off  2:off  3:off  4:off  5:off  6:off  7:off