recalbox/mk_arcade_joystick_rpi #6: Unable to load Driver

benduino, 2015-03-08:
Hi

it looks like there is a problem with the Raspberry PI 2.
Installing the Driver is just fine, but when i load the module:

sudo modprobe mk_arcade_joystick_rpi map=1
i get this error
ERROR: could not insert 'mk_arcade_joystick_rpi': Exec format error

i had no problems with the Raspberry PI B ( last week ) , but my new PI 2 wont work :(
i followed exactly your install procedure ... ( 3 time with new Retropie 2.6 image )

dmesg:
[ 1912.213882] mk_arcade_joystick_rpi: disagrees about version of symbol module_layout

modinfo mk_arcade_joystick_rpi:
filename:       /lib/modules/3.18.9-v7+/updates/dkms/mk_arcade_joystick_rpi.ko
license:        GPL
description:    GPIO and MCP23017 Arcade Joystick Driver
author:         Matthieu Proucelle
srcversion:     2B2E40AFD619744883AA3C0
depends:
vermagic:       3.18.9-v7 SMP preempt mod_unload modversions ARMv7
parm:           map:Enable or disable GPIO and MCP23017 Arcade Joystick (array of int)

uname -r
3.18.9-v7+

can you reproduce my Problem ?

regards
Beny

digitalLumberjack, 2015-03-08:
Hum weird.
The installation of the kernel headers returned no error ?

benduino, 2015-03-08:
you have to know i am a linux beginner ... sorry ;)

what i have done now:
wget http://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/linux-headers-`uname -r`_`uname -r`-2_armhf.deb
sudo dpkg -i linux-headers-`uname -r`_`uname -r`-2_armhf.deb

the i get 2 errors:

(Reading database ... 
(Reading database ... 5%
(Reading database ... 10%
(Reading database ... 15%
(Reading database ... 20%
(Reading database ... 25%
(Reading database ... 30%
(Reading database ... 35%
(Reading database ... 40%
(Reading database ... 45%
(Reading database ... 50%
(Reading database ... 55%
(Reading database ... 60%
(Reading database ... 65%
(Reading database ... 70%
(Reading database ... 75%
(Reading database ... 80%
(Reading database ... 85%
(Reading database ... 90%
(Reading database ... 95%
(Reading database ... 100%
(Reading database ... 74661 files and directories currently installed.)
Preparing to replace linux-headers-3.18.9-v7+ 3.18.9-v7+-2 (using linux-headers-3.18.9-v7+_3.18.9-v7+-2_armhf.deb) ...
make: Entering directory '/usr/src/linux-headers-3.18.9-v7+'
  CLEAN   .
scripts/Makefile.clean:17: drivers/infiniband/hw/amso1100/Makefile: No such file or directory
make[4]: *** No rule to make target 'drivers/infiniband/hw/amso1100/Makefile'.  Stop.
scripts/Makefile.clean:92: recipe for target 'drivers/infiniband/hw/amso1100' failed
make[3]: [drivers/infiniband/hw/amso1100] Error 2 (ignored)
scripts/Makefile.clean:17: drivers/infiniband/ulp/srp/Makefile: No such file or directory
make[4]: *** No rule to make target 'drivers/infiniband/ulp/srp/Makefile'.  Stop.
scripts/Makefile.clean:92: recipe for target 'drivers/infiniband/ulp/srp' failed
make[3]: [drivers/infiniband/ulp/srp] Error 2 (ignored)
scripts/Makefile.clean:17: drivers/scsi/cxgbi/cxgb3i/Makefile: No such file or directory
make[4]: *** No rule to make target 'drivers/scsi/cxgbi/cxgb3i/Makefile'.  Stop.
scripts/Makefile.clean:92: recipe for target 'drivers/scsi/cxgbi/cxgb3i' failed
make[3]: [drivers/scsi/cxgbi/cxgb3i] Error 2 (ignored)
scripts/Makefile.clean:17: drivers/scsi/cxgbi/cxgb4i/Makefile: No such file or directory
make[4]: *** No rule to make target 'drivers/scsi/cxgbi/cxgb4i/Makefile'.  Stop.
scripts/Makefile.clean:92: recipe for target 'drivers/scsi/cxgbi/cxgb4i' failed
make[3]: [drivers/scsi/cxgbi/cxgb4i] Error 2 (ignored)
scripts/Makefile.clean:17: drivers/scsi/osd/Makefile: No such file or directory
make[3]: *** No rule to make target 'drivers/scsi/osd/Makefile'.  Stop.
scripts/Makefile.clean:92: recipe for target 'drivers/scsi/osd' failed
make[2]: [drivers/scsi/osd] Error 2 (ignored)
scripts/Makefile.clean:17: drivers/usb/host/whci/Makefile: No such file or directory
make[4]: *** No rule to make target 'drivers/usb/host/whci/Makefile'.  Stop.
scripts/Makefile.clean:92: recipe for target 'drivers/usb/host/whci' failed
make[3]: [drivers/usb/host/whci] Error 2 (ignored)
scripts/Makefile.clean:17: fs/exofs/Makefile: No such file or directory
make[2]: *** No rule to make target 'fs/exofs/Makefile'.  Stop.
scripts/Makefile.clean:92: recipe for target 'fs/exofs' failed
make[1]: [fs/exofs] Error 2 (ignored)
fs/hostfs/Makefile:11: arch/um/scripts/Makefile.rules: No such file or directory
make[2]: *** No rule to make target 'arch/um/scripts/Makefile.rules'.  Stop.
scripts/Makefile.clean:92: recipe for target 'fs/hostfs' failed
make[1]: [fs/hostfs] Error 2 (ignored)
scripts/Makefile.clean:17: fs/nfs/objlayout/Makefile: No such file or directory
make[3]: *** No rule to make target 'fs/nfs/objlayout/Makefile'.  Stop.
scripts/Makefile.clean:92: recipe for target 'fs/nfs/objlayout' failed
make[2]: [fs/nfs/objlayout] Error 2 (ignored)
  CLEAN   .tmp_versions
[... 125 lines cut ...]

DKMS: uninstall completed.

------------------------------
Deleting module version: 0.1.4
completely from the DKMS tree.
------------------------------
Done.
Loading new mk_arcade_joystick_rpi-0.1.4 DKMS files...
First Installation: checking all kernels...
It is likely that 3.18.9-v7+ belongs to a chroot's host
Building for architecture armv7l
Building initial module for 3.18.9-v7+
Done.

mk_arcade_joystick_rpi:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/3.18.9-v7+/updates/dkms/

depmod....

DKMS: install completed.

oliver32, 2015-03-08:
I have the exact same problem, would really appreciate if somebody could find a solution :)

digitalLumberjack, 2015-03-08:
Maybe it's an issue with the kernel headers.
I would have done it myself but i won't have time this week, could anyone
try to install this driver for example. mk_arcade_joystick_gpio is based on
this one. If you manage to install the driver and load it (load it with
map=1 argument) the issue is on mk_arcade_joystick_rpi else it's about the
headers and i will tell marq's there is a bug with his header generation
script.

benduino, 2015-03-09:
where i can some find infos about : mk_arcade_joystick_gpio ?
google dont know mk_arcade_joystick_gpio ......

digitalLumberjack, 2015-03-09:
Sorry i was talking about mk_arcade_joystick_rpi.

oliver32, 2015-03-09:
I have tried to install the driver at least 4-5 times, with updated software, without latest software, latest kernel, without latest kernel. I always get the same message when I do: sudo modprobe mk_arcade_joystick_rpi map=1 and it says ERROR: could not insert 'mk_arcade_joystick_rpi': Exec format error

benduino, 2015-03-11:
ähhh ... now its working again :)
i spent the last 3 day's trying everything ... different sd images, compiling from source ...
then i had success with first installing the gamecon driver from retropie and then the mk_arcade_joystick_rpi. so it looks like the problem was with header files ...
so today i tray everything scratch ... with the normal install script and it worked !
somebody did something ;)

digitalLumberjack, 2015-03-12:
That is a good news :)

digitalLumberjack, 2015-03-12:
I'll wait for @oliver32 to test and report if it's ok for him, then i'll close this issue.

oliver32, 2015-03-13:
@digitalLumberjack everything is now working properly :)

digitalLumberjack, 2015-03-13:
Cool :)

closed 2015-03-13 by digitalLumberjack

bborncr, 2015-04-23:
I am experiencing the exact same error.  Using an RPI B. I made sure that I executed the reboot. I have done the procedure several times.

bborncr, 2015-04-23:
I will do the procedure later tonight and send the output.

bborncr, 2015-04-24:
Here is the section that starts to fail:
Installing current kernel headers
(Reading database ... 105576 files and directories currently installed.)
Preparing to replace linux-headers-3.18.7+ 3.18.7+-2 (using linux-headers-3.18.7+_3.18.7+-2_armhf.deb) ...
make: Entering directory '/usr/src/linux-headers-3.18.7+'
  CLEAN   .
scripts/Makefile.clean:17: drivers/infiniband/hw/amso1100/Makefile: No such file or directory
make[4]: *** No rule to make target 'drivers/infiniband/hw/amso1100/Makefile'.  Stop.
scripts/Makefile.clean:92: recipe for target 'drivers/infiniband/hw/amso1100' failed
make[3]: [drivers/infiniband/hw/amso1100] Error 2 (ignored)
scripts/Makefile.clean:17: drivers/infiniband/ulp/srp/Makefile: No such file or directory
make[4]: *** No rule to make target 'drivers/infiniband/ulp/srp/Makefile'.  Stop.
scripts/Makefile.clean:92: recipe for target 'drivers/infiniband/ulp/srp' failed
make[3]: [drivers/infiniband/ulp/srp] Error 2 (ignored)
scripts/Makefile.clean:17: drivers/scsi/cxgbi/cxgb3i/Makefile: No such file or directory
make[4]: *** No rule to make target 'drivers/scsi/cxgbi/cxgb3i/Makefile'.  Stop.
scripts/Makefile.clean:92: recipe for target 'drivers/scsi/cxgbi/cxgb3i' failed
make[3]: [drivers/scsi/cxgbi/cxgb3i] Error 2 (ignored)
scripts/Makefile.clean:17: drivers/scsi/cxgbi/cxgb4i/Makefile: No such file or directory
make[4]: *** No rule to make target 'drivers/scsi/cxgbi/cxgb4i/Makefile'.  Stop.
scripts/Makefile.clean:92: recipe for target 'drivers/scsi/cxgbi/cxgb4i' failed
make[3]: [drivers/scsi/cxgbi/cxgb4i] Error 2 (ignored)
scripts/Makefile.clean:17: drivers/scsi/osd/Makefile: No such file or directory
make[3]: *** No rule to make target 'drivers/scsi/osd/Makefile'.  Stop.
scripts/Makefile.clean:92: recipe for target 'drivers/scsi/osd' failed
make[2]: [drivers/scsi/osd] Error 2 (ignored)
scripts/Makefile.clean:17: drivers/usb/host/whci/Makefile: No such file or directory
make[4]: *** No rule to make target 'drivers/usb/host/whci/Makefile'.  Stop.
scripts/Makefile.clean:92: recipe for target 'drivers/usb/host/whci' failed
make[3]: [drivers/usb/host/whci] Error 2 (ignored)
scripts/Makefile.clean:17: fs/exofs/Makefile: No such file or directory
make[2]: *** No rule to make target 'fs/exofs/Makefile'.  Stop.
scripts/Makefile.clean:92: recipe for target 'fs/exofs' failed
make[1]: [fs/exofs] Error 2 (ignored)
fs/hostfs/Makefile:11: arch/um/scripts/Makefile.rules: No such file or directory
make[2]: *** No rule to make target 'arch/um/scripts/Makefile.rules'.  Stop.
scripts/Makefile.clean:92: recipe for target 'fs/hostfs' failed
make[1]: [fs/hostfs] Error 2 (ignored)
scripts/Makefile.clean:17: fs/nfs/objlayout/Makefile: No such file or directory
make[3]: *** No rule to make target 'fs/nfs/objlayout/Makefile'.  Stop.
scripts/Makefile.clean:92: recipe for target 'fs/nfs/objlayout' failed
make[2]: [fs/nfs/objlayout] Error 2 (ignored)
  CLEAN   .tmp_versions
ls: cannot access ./Documentation/DocBook/media/*.b64: No such file or directory

bborncr, 2015-04-24:
[   11.458193] mk_arcade_joystick_rpi: disagrees about version of symbol module_layout