recalbox/mk_arcade_joystick_rpi #7: Recompile with different pin numbers

Jimima, 2015-03-22:
Is there any chance you could provide instructions for recompiling the module from source? I want to be able to change the pin numbers as my joystick is already wired up in a different way :-/

Loxrie, 2015-05-04:
Once you've changed the pin numbers in the array you will need to modify the call on line 425 to match:

```
setGpioPullUps(0xBC6C610);
```

So note down the GPIO pins you are using in ascending order, then organise that right to left with a leading 0.  For example the above number corresponds to the pins for player 1 in the image on the github page:

PINS: 4,9,10,14,15,17,18,22,23,24,25,27

1011 1100 0110 1100 0110 0001 0000

The last zero on the right above is padding is padding,  So counting right to left we can see the position of the 1s correspond to the pins.  Take that binary and convert it to hex and you get 0xBC6C610.

Now my understanding the second line that calls the function is it's just setting all the pins to pull ups.  You may not want this behaviour, especially if like me you have some of the pins being used for other tasks and they don't want pulling up.  I think, though cannot yet confirm because of problems I've raised in #9 that you can modify that line too.  I think you'd need to change it to cover the pins you using for P1 and P2.  In my case that section of code reads:

```
        case MK_ARCADE_GPIO:
            for (i = 0; i < mk_max_arcade_buttons; i++) {
                setGpioAsInput(mk_arcade_gpio_maps[i]);
            }
            setGpioPullUps(0x9C6C01C);
            printk("GPIO configured for pad%d\n", idx);
            break;
        case MK_ARCADE_GPIO_BPLUS:
            for (i = 0; i < mk_max_arcade_buttons; i++) {
                setGpioAsInput(mk_arcade_gpio_maps_bplus[i]);
            }
            setGpioPullUps(0x40939E0 | 0x9C6C01C);
            printk("GPIO configured for pad%d\n", idx);
            break;
```

So I OR my two masks for the pins I use together.  So if I just configure 1 joystick I get the masks for P1, and if I configure 2 joysticks it ors the P1 and P2 masks together.

Hope that makes sense, and of course is accurate.  Hopefully the original developer can weigh in with any corrections for me.

Loxrie, 2015-05-04:
You can actually calculate the mask required from the array by iterating the array, bit shifting and using a logical OR.  I plan to write that code but can't really do so until I solve the issue I'm having.

E.g.
```
int32_t mask = 0x0000000;
static const int mk_arcade_gpio_maps[] = {4,  17,    27,  22,    10,    9,      25, 24, 23, 18, 15, 14 };
for (int i=0; i<12;i++) {
   int32_t pin_mask  = 1<<mk_arcade_gpio_maps[i];
   mask = mask | pin_mask;
}
```

above not testing but hopefully gives the idea.

roachman5000, 2016-02-16:
Hi, I actually have the same issue, as I cannot use GPIOs 14 & 15 on account of them being allocated for something else, but can basically use any of the GPIOs associated with a second joystick.  How would it go if I wanted to swap GPIOs 14 & 15 for GPIOs 7 & 8?  

Thanks so much in advance!

[image]

Loxrie, 2017-01-25:
I have no idea how I missed this comment for so long.  Looking through the code and the readme as it stands now the concept I presented would appear to be in there.  You can specify custom pin maps via module config now.  Sweet. 👍