recalbox/mk_arcade_joystick_rpi #12: Button 1 (btnSelect) & Button 5 (btnLThumb) Pressed

clodewyks, 2015-06-17:
As mentioned in the subject both buttons show "on" while running the test. Even if I remove all buttons from the GPIO pins it still shows that those two buttons are on. I have tried wiring dummy ground wires to those ports. What am I doing wrong or can test with?

clodewyks, 2015-06-17:
Edit: 1 = BtnY, 5 = BtnSelect

digitalLumberjack, 2015-06-17:
Hi, it must be an issue with the internal pullup settings. do you have an RPI2 ?

clodewyks, 2015-06-17:
Yes I do, B+. What can I do to change it?

digitalLumberjack, 2015-06-17:
RPI2 B+ ? B you mean ?

clodewyks, 2015-06-17:
Apologies, 2.

clodewyks, 2015-06-17:
Pi Model B+ V1.2, I'm sure that is the RPI2 B+ as that is what I ordered

clodewyks, 2015-06-18:
I've only noticed now that they might have sent me the wrong one. It is the Model B+, not RPI2. I checked and the wiring seems to be the same for both, what else can I try?