recalbox/mk_arcade_joystick_rpi #15: How to modify mk_joystick_arcade_rpi to change GPIO pins used on Raspberry Pi 2

jimi007, 2015-08-14:
Hi

I have recently setup a retropie 3 raspberry pi 2 box and have installed the mk_arcade_joystick_rpi module.
Working ok.
Only issue i'm having is that the GPIO joystick i am using needs to use specific GPIO pins on the Pi.

Is there anyway to recompile or change the GPIO pins being used in the MK_joystick module?

I am pretty new to linux so any help would be appreciated

thanks

Jimi

efraimsangil, 2015-11-03:
I have the some problem. I used GPIO pin 8 with Green LED for status of RPI. If blinking, I know that RPI it's working.
I tried to change the source code and compile, but always I show this error:

pi@retropie ~/mk_arcade_joystick_rpi $ sudo make
make -C /lib/modules/`uname -r`/build M= modules
make[1]: Entering directory '/usr/src/linux-headers-4.1.12-v7+'
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
make[2]: 'include/generated/mach-types.h' is up to date.
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
grep: scripts/../arch/x86/syscalls/syscall_32.tbl: No existe el fichero o el directorio
make[2]: *** No rule to make target 'arch/arm/crypto/aes-armv4.o', needed by 'arch/arm/crypto/aes-arm.o'.  Alto.
Makefile:947: recipe for target 'arch/arm/crypto' failed
make[1]: *** [arch/arm/crypto] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.1.12-v7+'
Makefile:9: recipe for target 'all' failed
make: *** [all] Error 2

¿What I doing wrong?

Regards,

digitalLumberjack, 2015-11-03:
Hi,
i may update the recalboxOS kernel to 4.X.X soon, so i'll take a look. For now, sorry i cannot help you more.

efraimsangil, 2015-11-03:
No problem! I can wait :)
Thanks!

gbraux, 2016-01-07:
Same problem here ... I suspect the 4.1.12-v7+ kernel headers to be a bit buggy (knowing that we can only get them from an untrusted source : https://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/ ...).

I may try to get generic headers instead ...

wermy, 2016-02-16:
Did anyone find a solution for this?  I'm in the exact same situation: need to change the pins and recompile, hitting the exact same error.  :\

digitalLumberjack, 2016-02-18:
Hi, we bumped to the 4.1.X kernel on recalbox, and everything is working well. For raspbian users, 
the kernel headers are really a pain to maintain and the recompilation of the kernel may be overkill too. 
Don't know if we could find a solution to compile the drivers easily...

roachman5000, 2016-03-17:
Hi, I have a similar issue.  I am running LiPoPi (or want to) as part of a portable pi zero gaming system running recalbox, and LiPoPi requires GPIOs 14 & 15.  Here is the link to the LiPoPi software and hardware setup:  https://github.com/NeonHorizon/lipopi

I also want to wire the controller buttons straight to the GPIOs in order to preserve the data USB port (Otherwise I'd wire the controller straight to that port and be done with it).  

For these reasons it would be simplest if I could wire into Player 2's GPIO layout and have the system recognize it as Player 1, since GPIOs 14 & 15 are mapped to Player 1 inputs.

Will it do that already?  Is there a way to swap Player 1 and Player 2 so Player 1's GPIO layout is switched with Player 2's?