recalbox/mk_arcade_joystick_rpi #18: ERROR : Unable to find kernel headers 4.1.10

rhobz, 2015-10-04:
Hey I just updated my retropie and cant get this to work after that.
It tries to download some 4.1.10-file but it is not found.

http://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/linux-headers-4.1.10-v7+_4.1.10-v7+-2_armhf.deb
Resolving www.niksula.hut.fi (www.niksula.hut.fi)... 130.233.43.7
Connecting to www.niksula.hut.fi (www.niksula.hut.fi)|130.233.43.7|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2015-10-04 19:45:45 ERROR 404: Not Found.
ERROR : Unable to find kernel headers

rhobz, 2015-10-06:
Works now, the file is available. GREAT!

closed 2015-10-06 by rhobz

digitalLumberjack, 2015-10-06:
Wouh an auto closable issue. I like that. I didn't answer directly cause i knew that marqs would have uploaded the headers quickly. Thanks to him.