recalbox/mk_arcade_joystick_rpi #23: Incorrect instructions regarding module options

joolswills, 2016-02-18:
Putting module options in /etc/modules won't work for systemd based setups (and /etc/modprobe.d is preferred on older systems also)

to fix on Jessie for example -

module name in /etc/modules

```
mk_arcade_joystick_rpi
```

and put the options in /etc/modprobe.d/mk_arcade_joystick.conf

```
options mk_arcade_joystick_rpi map=1,2
```

digitalLumberjack, 2016-02-18:
hi @joolswills 
thanks for the information i corrected that :)

For info is the mk_arcade_joystick_gpio in retropie ? Or are you using other drivers ?

closed 2016-02-18 by digitalLumberjack

joolswills, 2016-02-19:
We don't include it - but perhaps should as a few people have mentioned it recently. We include gamecon_gpio_rpi

digitalLumberjack, 2016-02-19:
ok nice :) 
tell me if you include the driver, we could work togother on the raspbian integration that is today hard to maintain due to the dependency on kernel headers.