recalbox/mk_arcade_joystick_rpi #29: Anybody had success with RP3 and this version of mk_arcade_joystick_rpi?

passion0721, 2016-04-17:
hi .
Recently I'm try to make arcade muchine whith Rasberry pi3 and 0.1.4 script and recalbox.
But here are the problem with below.

During run script, the recalbox Os not allow "apt-get" command.
    -> in my opionion, this symptom right situation but , how can i make further configuration?

can anybody check above problem?
thank you in advanced
best regards!

digitalLumberjack, 2016-04-17:
If you are using recalbox see : https://github.com/recalbox/recalbox-os/wiki/GPIO-controllers-(EN)

passion0721, 2016-04-17:
reply to umberjack

thank you for your kindly response. As your comment, I have to make configuration setting to enable.
thank you very much!!

closed 2016-04-17 by passion0721