recalbox/mk_arcade_joystick_rpi #32: FATAL : Module mk_arcade_joystick_rpi not found

oskarsjogren, 2016-05-08:
Hi.

I followed instructions to the last note. But when I try to load the driver with
`pi@raspberrypi:~/mkjoystick $ sudo modprobe mk_arcade_joystick_rpi map=1,2`

I get:

```
modprobe: FATAL: Module mk_arcade_joystick_rpi not found.
```

Is there something I can do about this issue? What is needed from your side to answer this question?

When I type
`lsmod`

I get:

```
Module                  Size  Used by
hidp                   13546  2
bnep                   10340  0
hci_uart               17943  1
btbcm                   5929  1 hci_uart
bluetooth             326105  9 bnep,hidp,btbcm,hci_uart
brcmfmac              186599  0
brcmutil                5661  1 brcmfmac
evdev                  11396  4
joydev                  9024  0
cfg80211              427855  1 brcmfmac
rfkill                 16037  3 cfg80211,bluetooth
snd_bcm2835            20511  1
snd_pcm                75698  2 snd_bcm2835
snd_timer              19160  1 snd_pcm
snd                    51844  4 snd_bcm2835,snd_timer,snd_pcm
bcm2835_gpiomem         3040  0
bcm2835_wdt             3225  0
uio_pdrv_genirq         3164  0
uio                     8000  1 uio_pdrv_genirq
uinput                  7454  0
i2c_dev                 5859  0
fuse                   83461  1
ipv6                  347530  34
```

No trace of mk_arcade_joystick_rpi... Any ideas on this?

Thank you

oskarsjogren, 2016-05-08:
An addition:

In /var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build/make.log I find the foolowing:

```
DKMS make.log for mk_arcade_joystick_rpi-0.1.4 for kernel 4.4.9-v7+ (armv7l)
Sun  8 May 16:30:53 UTC 2016
make -C /lib/modules/4.4.9-v7+/build M=/var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.9-v7+'
  CC [M]  /var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build/mk_arcade_joystick_rpi.o
In file included from include/linux/ktime.h:25:0,
                 from include/linux/rcupdate.h:47,
                 from include/linux/srcu.h:33,
                 from include/linux/notifier.h:15,
                 from include/linux/memory_hotplug.h:6,
                 from include/linux/mmzone.h:735,
                 from include/linux/gfp.h:5,
                 from include/linux/kmod.h:22,
                 from include/linux/module.h:13,
                 from /var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build/mk_arcade_joystick_rpi.c:31:
include/linux/jiffies.h:10:33: fatal error: generated/timeconst.h: No such file or directory
compilation terminated.
scripts/Makefile.build:264: recipe for target '/var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build/mk_arcade_joystick_rpi.o' failed
make[2]: *** [/var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build/mk_arcade_joystick_rpi.o] Error 1
Makefile:1384: recipe for target '_module_/var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build' failed
make[1]: *** [_module_/var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.4.9-v7+'
Makefile:9: recipe for target 'all' failed
make: *** [all] Error 2
```

rsa2016, 2016-05-25:
Hi i´ve got the same problem. Has someone a solution for that?

Thx

oskarsjogren, 2016-05-25:
Hi rsa2016.

I got rid of the problem when doing a fresh install. I had the issues when I had installed the retropie over a NOOBS system. Maybe the kernels are wrong, I have no idea. So if you have the possibility to download and burn the image directly, so so
https://retropie.org.uk/2016/04/retropie-3-7-is-released/
that worked for me anyway.

rsa2016, 2016-05-25:
Hi oskarsjogren,
thanks for your very fast answer. I used a fresh installation of raspbian image. I guess i try ist with the retropie-image.

rdubose, 2016-05-29:
Having the same issue.. It's a RaspberryPie 3 (seemed to have this working on a 2 at one point) using RetroPie 3.8 (also tried 3.6) have reloaded the image several times.

DKMS make.log for mk_arcade_joystick_rpi-0.1.4 for kernel 4.4.11-v7+ (armv7l)
Sun 29 May 05:23:55 UTC 2016
make -C /lib/modules/4.4.11-v7+/build M=/var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.11-v7+'
  CC [M]  /var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build/mk_arcade_joystick_rpi.o
In file included from include/linux/ktime.h:25:0,
                 from include/linux/rcupdate.h:47,
                 from include/linux/srcu.h:33,
                 from include/linux/notifier.h:15,
                 from include/linux/memory_hotplug.h:6,
                 from include/linux/mmzone.h:735,
                 from include/linux/gfp.h:5,
                 from include/linux/kmod.h:22,
                 from include/linux/module.h:13,
                 from /var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build/mk_arcade_joystick_rpi.c:31:
include/linux/jiffies.h:10:33: fatal error: generated/timeconst.h: No such file or directory
compilation terminated.
scripts/Makefile.build:264: recipe for target '/var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build/mk_arcade_joystick_rpi.o' failed
make[2]: *** [/var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build/mk_arcade_joystick_rpi.o] Error 1
Makefile:1384: recipe for target '_module_/var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build' failed
make[1]: *** [_module_/var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.4.11-v7+'
Makefile:9: recipe for target 'all' failed
make: *** [all] Error 2

jasonjennings, 2016-05-29:
I am experiencing the same problem as rdubose with a Raspberry Pi 3 B on RetroPie 3,8, I have reset it many times. I have not experienced this with RetroPie 3.7.

substring, 2016-05-29:
strangely enough, i thhink it compiles fine for recalbox with kernel 4.4
we arenot doing anything for retropie ... time to switch ? ;)
anyone checked if the missing timeconst.h is somewhere at least ?

jasonjennings, 2016-05-29:
Ah, I do like RetroPie, although I could look at RecalBox, worth a shot. If possible please do look into seeing if we can get a fix for 3.8 of retropie though, as it worked fine on 3.7.

simonster4, 2016-05-29:
I am experiencing the same issue.  I have tracked it down to something to do with the linux-headers package for 4.4.11 v7+ not compiling.  RetroPie 3.8 uses Kernel 4.4.11 where as 3.7 uses 4.1.19 which compiles fine.  I have also tried this with a Raspbian install and when updated to kernel 4.4.11, this doesn't install either, so it isn't RetroPie, but something to do with the Linux Headers files required by the GPIO Joystick.

I don't know enough about the headers files or how they work, but is it possible to build it using headers from another build?

substring, 2016-05-30:
looks like the `timeconst.h` is generated (at kernel compilation time ?) source : http://mrlaitaiyu.blogspot.fr/2015/01/how-to-compile-raspbian-on-raspbian.html
So, I'm afraid that is a retropie issue ... You might not copy/paste the header file from somewhere else because of the HZ value (it defines home much time is given to a process before switching to another).
Looks like the gamecon driver has the same problem

simonster4, 2016-05-30:
Over at RetroPie, Buzz has fix the gamecon driver issue:

"Looks like a problem with the external headers package. I noticed there is a packaged kernel headers file now included with Raspbian (I have a feeling there wasn't when this code was originally written), so I have adjusted the script to use that. I have also made it use the newer v1.2 version of gamecon-gpio and have fixed a minor version check bug I introduced recently."

simonster4, 2016-06-02:
I have resolved this issue for myself.  I followed the manual install process but instead step 2 (install linux headers), run:

sudo apt-get install raspberrypi-kernel-headers

This will install a headers package, however because I had been messing around I had to remove the old header files first, if you get a message about "file already in another package linux-headers-4.xxx.xxx" the run

sudo apt-get remove linux-headers-4.xx.xx

then rerun

sudo apt-get install raspberrypi-kernel-headers

once this has completed successfully, carry on with step 3a.

Not really understanding what I did, apart from muddling through, Is there any way to modify the install.sh to replace the headers install function?

jasonjennings, 2016-06-02:
simonster4, that did the trick!

ajefr, 2016-06-05:
@simonster4 
In fact, during step 1, a library header is installed.
In my case it was "linux-headers-3.6-trunk-common"
So you have to remove it using :
`sudo apt-get remove  linux-headers-3.6-trunk-common`
and then after install this one :
`sudo apt-get install raspberrypi-kernel-headers`
while done (it takes a while), you can go to step 3a

dav-m85, 2016-06-07:
@simonster4 was it a Pi3 in your case ? I just stumbled into another compilation error, so before I dig more...

jasonjennings, 2016-06-07:
Yes

simonster4, 2016-06-10:
Yes, mine is a Pi3

thrylax, 2016-06-19:
Thanks for the assist simonster4. Worked flawlessly. GPIO Joysticks now working perfectly on Raspberry Pi 3 running RetroPie v3.8.1.

Bobgee, 2016-08-03:
The instructions give the command to configure a single joystick, i.e. 

sudo modprobe mk_arcade_joystick_rpi map=5 gpio=pin1,pin2,pin3,.....,pin12

Does anybody know how to use custom GPIO for two joysticks?

Many Thanks

Bobgee, 2016-08-03:
Well it seems I spoke too soon with my question above, as I can't even get the single player customization to work.

I type:

sudo modprobe mk_arcade_joystick_rpi map=5 gpio=5,3,7,11,12,23,26,16,33,29,38,37
jstest /dev/input/js0

and no buttons works.

If I reboot and then type:

sudo modprobe mk_arcade_joystick_rpi map=1,2
jstest /dev/input/js0

and hot random buttons on my joystick I can see some of them register

I conlude something is wrong with my first modprobe line above but from what I can see it is correct as per the instructions.

Please, if anyone can help with this and also explain how to configure two pure gpio (no logic expander) joysticks, I would be extremely grateful.

Cheers

captainyesterdayy, 2017-04-15:
First and foremost thank you everyone for this and all the other opensource projects you all help with.

For anyone interested, this worked for me. On Retropie.

Had to resort to the manual steps like simonster4 did above.

I also discovered by trial and error that in the tests that I ran on different pre-built images out there that you do not need to run the update portion of the steps. Those are the longest part of the process. In some cases this took up to 1 hour to get done based on how out of date the pre-built image was.

Just for reference I will call them **optional** and are listed here for completeness skip down if you don't want to update your system:

Step 1 (Switch to root) - This took out any possible permissions issue.

`sudo su`

Step 2 (You will need to wait on this in front of your ssh session as you will need to hit "q" for a message prompt)

`apt-get upgrade -y`

Step 3 

`rpi-update -y`

**So these were the steps I took that worked for me (if you skipped the updates above):**

Step 1 (Switch to root) - This took out any possible permissions issue.

`sudo su`

Step 2 

`apt-get install -y --force-yes dkms cpp-4.7 gcc-4.7 git joystick`

Step 3

`apt-get install -y --force-yes raspberrypi-kernel-headers`

Step 4

`wget https://github.com/recalbox/mk_arcade_joystick_rpi/releases/download/v0.1.4/mk-arcade-joystick-rpi-0.1.4.deb`

Step 5

`dpkg -i mk-arcade-joystick-rpi-0.1.4.deb`

Step 6

`modprobe mk_arcade_joystick_rpi map=1,2`

Step 7 (I just combined the steps listed in the readme)

`echo 'mk_arcade_joystick_rpi' >> /etc/modules`

Step 8 (Same here)

`echo 'options mk_arcade_joystick_rpi map=1,2' > /etc/modprobe.d/mk_arcade_joystick.conf`

Step 9

`reboot`