recalbox/mk_arcade_joystick_rpi #35: Will not install with RetroPie 3.8 (Linux 4.4.11-v7+)

simonster4, 2016-05-29:
I have been using this since Christmas, and it was working great, but since updating to RetroPie 3.8 it will not install.  There appears to be issues compiling linux headers file.

I first noticed the issue after updating RetroPie from 3.7 to 3.8.  I then tryied using the RetroPie 3.8 image from their website, and it wont compile on this either.

Any help would be appreciated, as at the moment I have no controls for my RetroPie machine :-(

simonster4, 2016-05-29:
Here is the install messages:

Installing required dependencies
Reading package lists...
Building dependency tree...
Reading state information...
cpp-4.7 is already the newest version.
dkms is already the newest version.
gcc-4.7 is already the newest version.
git is already the newest version.
joystick is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Downloading current kernel headers
Installing current kernel headers
(Reading database ... 80411 files and directories currently installed.)
Preparing to unpack linux-headers-4.4.11-v7+_4.4.11-v7+-2_armhf.deb ...
make: Entering directory '/usr/src/linux-headers-4.4.11-v7+'
  CLEAN   .
scripts/Makefile.clean:86: recipe for target 'drivers/gpu/drm/nouveau' failed
scripts/Makefile.clean:86: recipe for target 'drivers/infiniband/ulp/srp' failed
scripts/Makefile.clean:86: recipe for target 'drivers/scsi/cxgbi/cxgb3i' failed
scripts/Makefile.clean:86: recipe for target 'drivers/scsi/cxgbi/cxgb4i' failed
scripts/Makefile.clean:86: recipe for target 'drivers/scsi/osd' failed
scripts/Makefile.clean:86: recipe for target 'drivers/staging/rdma/amso1100' failed
scripts/Makefile.clean:86: recipe for target 'drivers/usb/host/whci' failed
scripts/Makefile.clean:86: recipe for target 'fs/exofs' failed
scripts/Makefile.clean:86: recipe for target 'fs/hostfs' failed
scripts/Makefile.clean:86: recipe for target 'fs/nfs/objlayout' failed
  CLEAN   .tmp_versions
  CLEAN   scripts/basic
  CLEAN   scripts/dtc
  CLEAN   scripts/genksyms
  CLEAN   scripts/kconfig
  CLEAN   include/config include/generated arch/arm/include/generated
  CLEAN   .config Module.symvers
make: Leaving directory '/usr/src/linux-headers-4.4.11-v7+'
Unpacking linux-headers-4.4.11-v7+ (4.4.11-v7+-2) over (4.4.11-v7+-2) ...
Setting up linux-headers-4.4.11-v7+ (4.4.11-v7+-2) ...
make: Entering directory '/usr/src/linux-headers-4.4.11-v7+'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/basic/bin2c
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
*** Default configuration is based on 'bcm2709_defconfig'
#
# configuration written to .config
#
make: Leaving directory '/usr/src/linux-headers-4.4.11-v7+'
make: Entering directory '/usr/src/linux-headers-4.4.11-v7+'
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  WRAP    arch/arm/include/generated/asm/bitsperlong.h
  WRAP    arch/arm/include/generated/asm/cputime.h
  WRAP    arch/arm/include/generated/asm/current.h
  WRAP    arch/arm/include/generated/asm/emergency-restart.h
  WRAP    arch/arm/include/generated/asm/errno.h
  WRAP    arch/arm/include/generated/asm/exec.h
  WRAP    arch/arm/include/generated/asm/ioctl.h
  WRAP    arch/arm/include/generated/asm/ipcbuf.h
  WRAP    arch/arm/include/generated/asm/irq_regs.h
  WRAP    arch/arm/include/generated/asm/kdebug.h
  WRAP    arch/arm/include/generated/asm/local.h
  WRAP    arch/arm/include/generated/asm/local64.h
  WRAP    arch/arm/include/generated/asm/mm-arch-hooks.h
  WRAP    arch/arm/include/generated/asm/msgbuf.h
  WRAP    arch/arm/include/generated/asm/msi.h
  WRAP    arch/arm/include/generated/asm/param.h
  WRAP    arch/arm/include/generated/asm/parport.h
[... 41 lines cut ...]
It is likely that 4.4.11-v7+ belongs to a chroot's host
Building for architecture armv7l
Building initial module for 4.4.11-v7+
Downloading mk_arcade_joystick_rpi 0.1.4
Installing mk_arcade_joystick_rpi 0.1.4
(Reading database ... 80411 files and directories currently installed.)
Preparing to unpack mk-arcade-joystick-rpi-0.1.4.deb ...

------------------------------
Deleting module version: 0.1.4
completely from the DKMS tree.
------------------------------
Done.
Unpacking mk-arcade-joystick-rpi-dkms (0.1.4) over (0.1.4) ...
Setting up mk-arcade-joystick-rpi-dkms (0.1.4) ...
Loading new mk_arcade_joystick_rpi-0.1.4 DKMS files...
First Installation: checking all kernels...
It is likely that 4.4.11-v7+ belongs to a chroot's host
Building for architecture armv7l
Building initial module for 4.4.11-v7+
Error! Bad return status for module build on kernel: 4.4.11-v7+ (armv7l)
Installation OK
Load the module with 'sudo modprobe mk_arcade_joystick_rpi map=1' for 1 joystick
or with 'sudo modprobe mk_arcade_joystick_rpi map=1,2' for 2 joysticks
See https://github.com/digitalLumberjack/mk_arcade_joystick_rpi#loading-the-driver for more details

simonster4, 2016-05-29:
Opened the /var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build/make.log file and found this:

DKMS make.log for mk_arcade_joystick_rpi-0.1.4 for kernel 4.4.11-v7+ (armv7l)
Sun 29 May 20:37:11 UTC 2016
make -C /lib/modules/4.4.11-v7+/build M=/var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.11-v7+'
  CC [M]  /var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build/mk_arcade_joystick_rpi.o
In file included from include/linux/ktime.h:25:0,
                 from include/linux/rcupdate.h:47,
                 from include/linux/srcu.h:33,
                 from include/linux/notifier.h:15,
                 from include/linux/memory_hotplug.h:6,
                 from include/linux/mmzone.h:735,
                 from include/linux/gfp.h:5,
                 from include/linux/kmod.h:22,
                 from include/linux/module.h:13,
                 from /var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build/mk_arcade_joystick_rpi.c:31:
include/linux/jiffies.h:10:33: fatal error: generated/timeconst.h: No such file or directory
compilation terminated.
scripts/Makefile.build:264: recipe for target '/var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build/mk_arcade_joystick_rpi.o' failed
make[2]: *** [/var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build/mk_arcade_joystick_rpi.o] Error 1
Makefile:1384: recipe for target '_module_/var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build' failed
make[1]: *** [_module_/var/lib/dkms/mk_arcade_joystick_rpi/0.1.4/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.4.11-v7+'
Makefile:9: recipe for target 'all' failed
make: *** [all] Error 2

substring, 2016-05-29:
You are missing a dependancy on a header file : timeconst.h. Dunno how retropie ships everything, but you miss a dev lib thay you should be able to install with apt-get

simonster4, 2016-05-29:
It appears to be related to the kernel update.

When I run the mkjoystick/install.sh, it attempts to install 
https://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-4.4.10-v7+_4.4.10-v7+-2_armhf.deb
which fails on creating the timeconst file.  It used to work before the kernel update and I have confirmed that this is the only thing that can be affecting it.  Is there any way to use a different set of linux headers?

substring, 2016-05-29:
dupe of #32 
can you close and post further infos there ?

closed 2016-05-29 by simonster4