recalbox/mk_arcade_joystick_rpi #37: Ingame problems

Augment1n, 2016-06-10:
Hi! I installed rretropie, did all your stuff, successfully configurated and tested the 2 player's joysticks and buttons, but in the games they are not working. 

Could you help me?

Thanks in advance.

obossio, 2016-06-10:
For snes, n64? can u be more specific? How do you connect the joystick?

Augment1n, 2016-06-10:
snes, megamen. I did everything as you said in the README.md.

Augment1n, 2016-06-10:
and in the Doom too. :(

Augment1n, 2016-06-10:
I assume i missed a setting in the retropie, just don't know what and where can I find it.

obossio, 2016-06-10:
but the menu works, you can go left and rigth?

Augment1n, 2016-06-10:
Yes. Both of the joysticks and all the 16 buttons are working in the menu.

Augment1n, 2016-06-10:
Could you tell me what are the right values?

#Player 1
input_player1_joypad_index = 0
input_player1_b_btn = ?
input_player1_a_btn = ?
input_player1_y_btn = ?
input_player1_x_btn = ?
input_player1_l_btn = ?
input_player1_r_btn = ?
input_player1_start_btn = ?
input_player1_select_btn = ?
input_player1_up_btn = ?
input_player1_down_btn = ?
input_player1_left_btn = ?
input_player1_right_btn = ?
input_exit_emulator_btn = ?
input_menu_toggle_btn = ?
#
#Player 2
input_player2_joypad_index = 1
input_player2_b_btn = ?
input_player2_a_btn = ?
input_player2_y_btn = ?
input_player2_x_btn = ?
input_player2_l_btn = ?
input_player2_r_btn = ?
input_player2_start_btn = ?
input_player2_select_btn = ?
input_player2_up_btn = ?
input_player2_down_btn = ?
input_player2_left_btn = ?
input_player2_right_btn = ?

I used this:
[image]

Augment1n, 2016-06-11:
Or how can I tell the retroarch to use js0 as player 1 and js1 as player 2?

Augment1n, 2016-06-11:
okey, i find where can I modify the RetroArch joypad buttons. It is working now

lemmons11, 2017-06-07:
Hey Augment1n can you share your fix? I am having the same issue.