recalbox/mk_arcade_joystick_rpi #38: Custom gpio pins no longer working?

a-dekker, 2016-06-11:
Custom gpio pins no longer working?

I can no longer get my gpio based joystick to work properly. It used to work fine (after reassigning buttons) with my retropie installation on a RPI2, but in later releases it fails. Not sure if it is related to a more recent kernel. The issue is the same on recallbox.

The gpio pins I use are custom. Running a python test program tells me the arcade joystick still reacts fine on the expected pins.

Loading the kernel module (sudo modprobe mk_arcade_joystick_rpi map=1) and running jstest on js0 makes it recognize the directions, but only 2 of the 8 buttons.
Trying to define custom gpio pins (sudo modprobe mk_arcade_joystick_rpi map=5 gpio=pin_nr,..) results in not a single move or button is recognized using jstest.

My gpio config:
```
right    pin 11
left     pin 7
ground   pin 9
up       pin 15
down     pin 13
button_1 pin 19
button_2 pin 21
button_3 pin 23
button_4 pin 29
button_5 pin 31
button_6 pin 33
button_7 pin 35
button_8 pin 37
```
Can't recall I did something extra in the past to make it work. Tested with the latest Retropie (v3.8.1) and recalboxOS (4.0.0-beta3).

a-dekker, 2016-06-25:
Manually editing the gpio_maps in mk_arcade_joystick_rpi.c did help.

riskpw, 2016-06-29:
i solved gpio controller not change.
here my github : https://github.com/rasplay/mk_arcade_joystick_rpi

a-dekker, 2016-06-29:
So did you actually change something to make it work? All I can find is a version change and some readme updates. mk_arcade_joystick_rpi.c is identical to upstream as far as I can tell (which I used).

riskpw, 2016-06-29:
our post is write can source chage and recompile mk_arcade_joystick_rpi.c  file.

```
// Map of the gpios :                     up, down, left, right, start, select, a,  b,  tr, y,  x,  tl 
static const int mk_arcade_gpio_maps[] = {4,  17,    27,  22,    10,    9,      25, 24, 23, 18, 15, 14 }; 
// 2nd joystick on the b+ GPIOS                 up, down, left, right, start, select, a,  b,  tr, y,  x,  tl 
static const int mk_arcade_gpio_maps_bplus[] = {11, 5,    6,    13,    19,    26,     21, 20, 16, 12, 7,  8 }; 
// Map of the mcp23017 on GPIOA            up, down, left, right, start, select 
static const int mk_arcade_gpioa_maps[] = {0,  1,    2,    3,     4,     5      }; 
 
// Map of the mcp23017 on GPIOB            a, b, tr, y, x, tl 
 static const int mk_arcade_gpiob_maps[] = {0, 1, 2,  3, 4, 5 }; 
 
// Map joystick on the b+ GPIOS with TFT      up, down, left, right, start, select, a,  b,  tr, y,  x,  tl 
static const int mk_arcade_gpio_maps_tft[] = {21, 13,    26,    19,    5,    6,     22, 4, 20, 17, 27,  16 }; 
```
if, you want to anyone kind of raspberrypi gpio. 

and can recompile file.

ours not chage recalbox team source. 
but write it that how to recompile recalbox team mk_arcade_joystick_rpi.c file.

a-dekker, 2016-06-29:
In that case you did the same as I did as I mentioned earlier: editing the gpio_maps in mk_arcade_joystick_rpi.c (and compiling).
I was hoping to be able remap without changing the code and recompiling (using map=5 gpio=pin_nr etc), but that did not work.

riskpw, 2016-06-29:
ours recompile 0.1.6 ver thought 01.4 ver. and raspberrypi gpio test finished.
finally,success gpio remapping on the retropie 3.8.1 and 4.0 beta