recalbox/mk_arcade_joystick_rpi #40: I can't get the custom GPIO map to work

Bobgee, 2016-08-05:
Hi, not sure what I am doing wrong here.

I built a dual arcade joystick with a raspberry pi header sometime ago and don't really want to completely re-wire it becuase I use it for other projects too. Ideally I would like to remap both joysticks but if I can get one working that would be a nice start.

I managed (after suffereing many problems with kernel headers) to get everything installed ok and if I use:

sudo modprobe mk_arcade_joystick_rpi map=1,2
jstest /dev/input/js0

and hit random buttons, I see them register.

If I reboot my pi and then try:

sudo modprobe mk_arcade_joystick_rpi map=5 gpio=5,3,7,11,12,23,26,16,33,29,38,37
jstest /dev/input/js0

None of the buttons do anything. 

From the instructions this seems to be the correct command, but I can't see what I have done wrong. Any ideas?

Also, I couldn't figure out from the instructions how to customize a dual joystick system, i.e. how to I customize two maps for JS0 and JS1, is that possible or do I have to rewire my joystick and use map1,2?

I should also add for clarity, I don't use any logic expander, just two standard joystick and a bunch of buttons connected directly to valid GPIO pins.

Cheers

substring, 2016-08-05:
I'm not sure we support yet gpio mapping. @digitalLumberjack could confirm. Or maybe it's the hotkey branch that doesn't support gpio mapping, can't remember

Pinuct, 2016-08-05:
In the current release of recalbox they are using the hotkey branch, that have not any gpio customization. I made the merge with the hotkey branch and also the gpio customization for both players. The pull request is made and now it is in revision for checking everything is fine. So i hope it will be aviable soon.

Bobgee, 2016-08-06:
Thanks for the help. Can I check my understanding with you.

You are in the process of updating the release and then I will be able to download latest version, install and then set gpio customisation for two joysticks?

Will I just have to repeat step 3a:

wget https://github.com/digitalLumberjack/mk_arcade_joystick_rpi/releases/download/v0.1.4/mk-arcade-joystick-rpi-0.1.4.deb
sudo dpkg -i mk-arcade-joystick-rpi-0.1.4.deb

but the version will be different?

Also, can you please tell me the command for customizing two joysticks as the instruction only explains for one?

Many many thanks!

substring, 2016-08-06:
@Pinuct submitted some patches to the repo to improve the current driver. Only @DigitalLumberjack has rights to merge his code and he's on holidays atm. It's a matter of days/few weeks.

Bobgee, 2016-08-06:
Ok thanks,

Pinuct, 2016-08-06:
By the moment you can't try it in my [repository](https://github.com/Pinuct/mk_arcade_joystick_rpi/tree/customgpiohotkey)

The instructions for install the module and configure the 2 players are in the readme.

If you find some errors pls tell us.

Thanks.

Bobgee, 2016-08-07:
Pinuct,

Thanks very much for your help.

Should I assume I must follow 3b instead of 3a? 

I have already installed the standard v0.1.4 using the 3a "install from release method".

Will there be any conflict, do I have to uninstall the other version? If so, how?

Thanks again

Pinuct, 2016-08-09:
You have to follow the 3b.

For removing a module use:
dkms remove <module_name> --all

Bobgee, 2016-08-10:
Thanks very much, I managed to uninstall the old version with the following:

sudo dkms remove -m mk_arcade_joystick_rpi -v 0.1.4 --all

It said "no origional module found for this module on this kernal" but went on to say it has successfully removed completely from the DKMS tree. I had all kinds of trouble with the kernal headers first time around, but I am confident I properly installed v0.1.4 eventually and I had it working. 

Anyway, I am sure I uninstalled it ok and sudo modprobe would not find the module after the remove.

The above is just for info if anyone else having trouble, I now have a fresh problem.

I am falling at the first hurdle of the 3b install route. 

I type 

git clone https://github.com/Pinuct/mk_arcade_joystick_rpi/tree/customgpiohotkey

I get:

cloning in to 'customgpiohotkey'...
FATAL repository 'https://github.com/Pinuct/mk_arcade_joystick_rpi/tree/customgpiohotkey' not found

Can you see what I am doing wrong? 

Also, later on in the instructions it says:

3.b.3 - Copy the files into the folder:

cd mk_arcade_joystick_rpi/

I am curious, where is mk_arcade_joystick_rpi ? Do I need to create this directory first and if so, does it matter where I do it?

Really apreciate your help!!!

Bobgee, 2016-08-10:
Ok, I spent a little while looking at this and realised that the git clone command does not like the /tree/xxx part.

I looked at the differences between 

https://github.com/Pinuct/mk_arcade_joystick_rpi/tree/customgpiohotkey

and

https://github.com/Pinuct/mk_arcade_joystick_rpi/

and found only the mk_arcade_joystick_rpi.c and readme files are recent.

So, I used git clone for the mk_arcade_joystick_rpi repository and used wget to get the .c file from the customgpiohotkey branch. I removed the .c from the git clone directory and replaced it with the version from the customgpiohotkey branch.

I then completed the instructions and it failed the build command.

I got:

dkms.conf:Error! No 'package_version' directive specified.
Error! Bad conf file.
File:/usr/src/mk_arcade_joystick_rpi-0.1.5/dkms.conf
Does not represent a valid dkms.conf file.

Pinuct, 2016-08-10:
open dkms.conf and replace:
    `PACKAGE_VERSION="$MKVERSION"`
with:
    `PACKAGE_VERSION="0.1.5"`

Bobgee, 2016-08-10:
Thanks again,

I tried that and got the following error messages:

Bad return status for module build on kernal: 4.4.13-v7+ (armv71)

make kernelrelease=4.4.13-v7+ all kversion=4.4.13-v7+.....(bad exit status:2)

I had problems with kernel headers initially but I thought I had resolved them becuase I was able to install the standard version of mk_arcade_joystick_rpi using step 3a method.

Basically from start, I did the following on a raspberry pi 3:

installed 3.8.1 retro pie, ran the update from within the options menu. then followed steps from:
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install rpi-update
sudo rpi-update
sudo apt-get install -y --force-yes dkms cpp-4.7 gcc-4.7 git joystick
sudo apt-get install raspberrypi-kernel-headers

Note from the text it appears that the headers installed were 4.4.13+ and 4.4.13-v7+ but the kernel was 4.4.15-v7+

I then downgraded the kernel to 4.4.13-v7+ using

sudo rpi-update "long hash tag"

after this I could install the standard module.

Any advice would be most welcome, thank you!

Pinuct, 2016-08-12:
i can't help you much but try to use a fresh image, intall it again and see if the problem still occurs. 
#41 had a kernel problem too, i don't know if it's related but he uses the new kernel command (the one that is in the pull request)  to install the kernel headers and it works for him.

Bobgee, 2016-08-12:
thanks for the help, 

basically I have tried that command, that is how I installed my headers although I kind of did it the other way around. 

I used that command and it still installed an older header, then I downgraded my kernel to match the installed headers. 

It fixed teh problem for installed 0.1.4 using the 3a method, so what I did with the headers, was enough to install the main version, but not your branch. 

I don't know what else to do other than to re-wire my joystick and use the standard version.

I really hoped with some effort I could install your version, but sadly it looks like I won't get it to work.

Thanks anyway for all your help.

VillemusPrime, 2016-10-04:
I'm actually having errors compiling. I get the following: 
`$ dkms build -m mk_arcade_joystick_rpi -v 0.1.5
dkms.conf: Error! No 'PACKAGE_VERSION' directive specified.
Error! Bad conf file.
File: /usr/src/mk_arcade_joystick_rpi-0.1.5/dkms.conf does not represent a valid dkms.conf file.`

Sorry for the formatting, I couldn't figure out how to get it to wrap correctly. Can anyone help me see what I'm doing wrong?

Pinuct, 2016-10-04:
Try changing in file dkms.conf:
`PACKAGE_VERSION="$MKVERSION"`
with:
`PACKAGE_VERSION="0.1.5"`

VillemusPrime, 2016-10-05:
Thanks, but I'm getting the same error...

`$ dkms.conf: Error! No 'PACKAGE_VERSION' directive specified.
Error! Bad conf file.
File: /usr/src/mk_arcade_joystick_rpi-0.1.5/dkms.conf
does not represent a valid dkms.conf file.`

bleem313, 2017-04-13:
Any updates? I'm getting the same errors as VillemusPrime

StoneSet, 2018-01-21:
Me, i get this error :
kernel source for this kernel does not seem to be installed.
any suggestion ?

BenKr, 2020-01-12:
> Thanks, but I'm getting the same error...
> 
> `$ dkms.conf: Error! No 'PACKAGE_VERSION' directive specified. Error! Bad conf file. File: /usr/src/mk_arcade_joystick_rpi-0.1.5/dkms.conf does not represent a valid dkms.conf file.`

Are you editing the correct file?
The file to edit is this one (if you are following the guide and using 0.1.5):
/usr/src/mk_arcade_joystick_rpi-0.1.5/dkms.conf