recalbox/mk_arcade_joystick_rpi #41: Unable to build on retropie (4.4.13-v7+)

Jeroen-R, 2016-08-11:
Executing install.sh results in an error on the following line:

`wget http://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/linux-headers-`uname -r`_`uname -r`-2_armhf.deb`

This is basically because http://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/linux-headers-4.4.13-v7+_4.4.13-v7+-2_armhf.deb does not exist.

I was able to make it work by replacing that line with the following:

`sudo apt-get install raspberrypi-kernel-headers`

I have no idea about how this change will affect other Linux versions, but it is worth considering implementing this change.

Pinuct, 2016-08-12:
Is implemented in #39

Jeroen-R, 2016-08-12:
Unless I am mistaken (new to github) that updates the readme, but not the install.sh file.

Pinuct, 2016-08-14:
You are right, it is not in the install.sh. But there are some problems with newer kernel versions (4.4.15-v7+) #40 and using that command don't solve it.