recalbox/mk_arcade_joystick_rpi #43: Other pin / button configuration. recompile?

tarifari, 2016-10-01:
Please fill the following form :
**What happened**

Wanted to use this recalbox i2c driver to work with my kitsch bent i2c dmg button pcb. (http://store.kitsch-bent.com/product/expanded-dmg-button-pcb-i2c-version)

**What should have happened**
Thought the configuration would work like expacted, but checked the pinout and found out, that this driver is using an another pinout configuration.

This is the pinout, used by the kitsch bent i2c controller:
pinout:

GPA2 — B
GPA3 — A
GPA4 — R2
GPA5 — R1
GPA6 — X
GPA7 — Y
GPB0 — UP
GPB1 — L1
GPB2 — L2
GPB3 — LEFT
GPB4 — RIGHT
GPB5 — DOWN
GPB6 — SELECT
GPB7 — START

this driver is running with other pinouts, which concludes in wrong / not working buttons.

Is it possible to recompile the driver, for use with the pinout above?

tried it by myself, but couldnt make it to a functional version.

tarifari, 2016-10-05:
no one with an idea?

Turael, 2016-10-09:
Hey tarifari,
i have the same problem but you are able to change the pin in this driver. 
See https://github.com/recalbox/mk_arcade_joystick_rpi/blob/master/mk_arcade_joystick_rpi.c#L186
and
https://github.com/recalbox/mk_arcade_joystick_rpi/blob/master/mk_arcade_joystick_rpi.c#L189

after edit this file you will need to recompile...

But i´ve got another problem: my start button is not working (GPB 7) - all other buttons are working...

d-rez, 2017-05-25:
It's fixed in master. Make sure to compile from source from the master branch. Use '''utils/custombuild''' for how to remove old modules and edit it to build and install from scratch. 

Worked for me