recalbox/mk_arcade_joystick_rpi #44: Map=4, Map=5 arguments not functioning

VillemusPrime, 2016-10-02:
As the title says, I can't get these arguments to work. They will load without error, but nothing works. 

I ran jstest, and EVERY button input was showing as "on". When I try "Map=1", or "Map=1,2", jstest runs fine every time I click a button.

Please let me know if I need to provide more details.

ian57, 2016-10-02:
Did you compile the module by yourself? If you use the 4.0 or 4.1 recalbox, they still use the old 1.05 release which do not include the new mapping. You need to use the "master" branch from the git hub

VillemusPrime, 2016-10-02:
It was compiled. I followed the installation instructions on the main page, with the exception of installing the raspberrypi-kernel-headers, as recommended in #32 

Also, I just noticed at the bottom of Issue #32 that user @Bobgee is having the same issue that I'm having.

ian57, 2016-10-03:
Sorry I cannot help you for retropie support. I only use Recalbox.

d-rez, 2017-05-25:
Make sure to compile from source from the master branch. 'Use utils/custombuild' for how to remove old modules and edit it to build and install from scratch. 

Worked for me