recalbox/mk_arcade_joystick_rpi #46: unable to load driver if self compiled

t4rif4ri, 2016-10-06:
Does anyone have the same problem?

If i run the automatic installation i get a working driver, but the pin configuration is wrong cause i use the kitsch bent i2c pcb board.
So i tried to do the manual installation and use the custom pinout. Installation is running well through, but i am unable to run the driver.

Dont know why. Maybe because of dkms instead of opkg?

vizigr0u, 2017-03-31:
I submitted a [pull request](https://github.com/recalbox/mk_arcade_joystick_rpi/pull/52) that fixes the bug with MCP23017 not initializing (I was trying to get the kitsch-bench i2c board to work as well ;) )

t4rif4ri, 2017-03-31:
Hey, i got mine now working with this driver

https://github.com/soundsrc/joypi

Maybe it could help you too. Good luck

vizigr0u, 2017-03-31:
I guess I was too stubborn to look for another driver, I'm glad I got it
working. thanks for the tip though! :)