recalbox/mk_arcade_joystick_rpi #53: Analog Joystick With arduino?

ArnavJain13, 2017-05-11:
I Have 2 Analog Joysticks which have (2 pots and a button) i would like to use with this library on a raspberry pi 2 (Recalbox). I know i cant directly connect them they need an adc which is where i would like to use an arduino pro mini,  to take the signal from the joysticks and interface it with the raspberry pi.
the problem is i don't know the procedure to interface the arduino and the pi.
Is it possible to do a digitalwrite(HIGH) from the arduino to the specified gpio on the raspberry pi (With a voltage divider setup to change the voltage level to 3.3v for the rasp)?
or do i have to connect them through serial and somehow change the code to get values from the serial connection

substring, 2017-05-12:
You can purchase some analog encoders like the A-Pac from Ultimarc. That's, I guess, your very best solution ...

> Is it possible to do a digitalwrite(HIGH) from the arduino to the specified gpio on the raspberry pi (With a voltage divider setup to change the voltage level to 3.3v for the rasp)?

The real problem behind is that the driver can't handle analog input, even if you could wire everything up.
Say each axis is encoded in 8 bits, which means 8 GPIO per axis, so 16 gpio for a full axis. You could plug just one joystick on the GPIO. 

In such case, I think using I2C would be "better" but the programming problem is still the same : this driver wasn not coded with analog in mind.

ian57, 2017-05-12:
@ArnavJain13 there is a simpler solution for that. A small eletronic device  with some resistors and 1 comparator. I made that to use PS2/3 analogue joystick on my nano cabinet. 
[image]

ArnavJain13, 2017-05-13:
Thanks @substring. And @ian57 for the quick answers, just what i needed, i figured out the arduino part but i guess using lm339n is more efficient. 
Im closing the thread

closed 2017-05-13 by ArnavJain13