recalbox/mk_arcade_joystick_rpi #71: recalbox 6.0-DragonBlaze  /recalbox  4.1 gpio buttons do not work at zero but work at pi3

DRAgon734465502, 2019-05-21:
I have try mk_arcade_joystick_rpi at the newest recalbox  4.1

I active the GPIO buttons by this code  

nano /recalbox/share/system/recalbox.conf

controllers.gpio.enabled=1
controllers.gpio.args=map=5 gpio=5,6,16,13,26,19,21,20,14,12,15,23,3

this work at pi3/3B+  but do not work at zero/zero W 
anyone have some problem ?? 

thinks

ian57, 2019-05-21:
what are the symptoms ? what does evtest reply? Be careful if you activated sounds on the gpio.. gpio 13 is used for PWN sound output because pizero has no analog output for sound. 

Please tell us more .

DRAgon734465502, 2019-05-21:
Yes I use PWM audio out 
nano /boot/config.txt
dtoverlay=pwm

I have enabled GPIO LCD and can display OK 
LCD code
nano /boot/config.txt
dtparam=spi=on
dtoverlay=pwm
hdmi_force_hotplug=1
hdmi_cvt=480 480 60 1 0 0 0
hdmi_group=2
hdmi_mode=87

nano /etc/modules.conf
flexfb 
fbtft_device
nano /etc/modprobe.d/fbtft.conf
options flexfb setaddrwin=0 buswidth=8  width=240 height=240 init=-1,0x11.. -1,0x29,-3
options fbtft_device debug=3 bgr=1 fps=60 custom=1rotate=90 name=flexfb speed=64000000 gpios=reset:27,dc:25,led:24

nano /etc/init.d/rcS
fbcp &

Then GPIO buttons config at recalbox.conf
system.fbcp.enabled=0

controllers.gpio.enabled=1
controllers.gpio.args=map=5 gpio=5,6,16,13,26,19,21,20,14,12,15,23,3

global.videomode=default 
global.ratio=1/1

DRAgon734465502, 2019-05-21:
this is all code , which work fine at PI3 

But at zero, jsut display and audio work ,  the GPIO buttons do not work, 
press the buttons do not work

ian57, 2019-05-21:
First : which release of Recalbox?
second : try to deactivate pwm to see if it is in cause. comment #dtoverlay=pwm in confit.txt

DRAgon734465502, 2019-05-21:
recalbox 6.0-DragonBlaze 
uname -a 
show linux  4.1

DRAgon734465502, 2019-05-21:
active the GPIO buttons by this code

nano /recalbox/share/system/recalbox.conf
controllers.gpio.enabled=1
controllers.gpio.args=map=5 gpio=5,6,16,13,26,19,21,20,14,12,15,23,3

which will make recalbox starting first desktop

DRAgon734465502, 2019-05-21:
Do not add that code , it is like this picture
http://ae01.alicdn.com/kf/HTB1dlCnXRWD3KVjSZKPq6yp7FXan.jpg

after adding  that code , it is like this picture
http://ae01.alicdn.com/kf/HTB1M_KnXSWD3KVjSZSgq6ACxVXab.jpg

DRAgon734465502, 2019-05-21:
I do not why the first starting have been changed from favorites to AMSTRRAD  when active the gpio buttons

ian57, 2019-05-21:
are the controls working properly ? did you try to deactivate pwm?

DRAgon734465502, 2019-05-21:
I do not active PWM audio output and GPIO LCD and just active GPIO buttons ,
Then I find  the first starting have been changed from favorites to AMSTRRAD
After I  disable GPIO buttons, the  first starting have gone back to favorites

DRAgon734465502, 2019-05-21:
controls  all do not work ,  I have tried to  activate pwm  or deactivate pwm . same result 

I also change two raspberry pi zero , same ..

DRAgon734465502, 2019-05-21:
the PI3 iamge and zero image from here 
https://archive.recalbox.com/

PI3 can work fine . 
zero will work like that . 

which  is bug at recalbox for pi0 ???

ian57, 2019-05-21:
ok seems to be a bug. need to make some test on my own to confirm.

DRAgon734465502, 2019-05-21:
after I change controllers.gpio.args=map=5 gpio=5,6,16,13,26,19,21,20,14,12,15,23,3
to                   controllers.gpio.args=map=4  .. 

make "5" to "4" the first starting have been changed to this automatically
http://ae01.alicdn.com/kf/HTB1VUqpXQ9E3KVjSZFrq6y0UVXav.jpg

But  all gpio controls  still do not work

DRAgon734465502, 2019-05-21:
which show "gamepad 1" "input require"
but the app can not check gpio buttons

ian57, 2019-05-21:
do not try 4, custom gpio is 5 and 6 that's all. if gpio a detected uou shoud see "custom GPIO"

ian57, 2019-05-23:
Nope sorry too much work, no time to test.

DRAgon734465502, 2019-05-23:
OK , I will wait your news , Thanks

Nickrash, 2019-05-29:
HI! I have the same problem, and i couldn´t fix it yet...

DRAgon734465502, 2019-05-30:
Yes . I find this problem also 

Now I still do not know how to fix this problem

nsilveri, 2019-05-31:
i have the same problem with custom gpio buttons

ian57, 2019-06-01:
Hi there, the bug is confirmed.... The bug seems to come from my code... I made a mistake in the PERI_BASE address. 
I will  recompile the module with the corrected code, test it and let you know.
This will be good for 6.1 I hope, Will put the module for downloadinf somewhere for the brave ;)
Thanks a lot for the feeback.

ian57, 2019-06-02:
Hi there ... good news, correcting the code fixes the bug. For the brave you can DL the new mk_arcade_joystick_rpi.ko at https://mega.nz/#!y9kCEI7I!7HdyKzhWS4HhfVgpvj7SM9RbnLwMUlGXoUqcxzuFCQM
replace the /lib/modules/4.14.62/extra/mk_arcade_joystick_rpi.ko with the new one and enjoy :)

for the others... please wait the 6.1 release..; sorry

DRAgon734465502, 2019-06-03:
good news ,  thanks

DRAgon734465502, 2019-06-03:
I will test it soon

Nickrash, 2019-06-05:
Hi! I already test it, but I have an issue. Now it recognize me a button but an incorrect one. When I push down, it goes up, is the only button that works.
Is my configuration of gpio correct? 
I used this one:
[image]

Regards.

ian57, 2019-06-05:
I've tested the Map=1,2 successuly on my pi0. I think tout should verify your wiring.

Nickrash, 2019-06-05:
I changed the Map=1 to Map=1,2 in recalbox.conf and still not working.