recalbox/mk_arcade_joystick_rpi #73: setup_timer needs to be replaced by timer_setup

phil123456, 2019-08-06:
Hello,

I tried to compile the driver but apparently setup_timer is deprecated

`
/home/pi/driver/driver.c: In function ‘mk_probe’:
/home/pi/driver/driver.c:524:5: error: implicit declaration of function ‘setup_timer’; did you mean ‘del_timer’? [-Werror=implicit-function-declaration]
     setup_timer(&mk->timer, mk_timer, (long) mk);
     ^~~~~~~~~~~
     del_timer
`

how can I fix this ?
I tried using timer_setup but the parameters are way different

thanks

brunopinheiro, 2019-08-15:
https://github.com/recalbox/mk_arcade_joystick_rpi/issues/62

The changes described here worked fine for me.

jclauzel, 2019-10-28:
Made a fork to fix this here: https://github.com/jclauzel/mk_arcade_joystick_rpi the following should work:
sudo git clone https://github.com/jclauzel/mk_arcade_joystick_rpi.git