recalbox/mk_arcade_joystick_rpi #75: Can I use this driver at raspberry pi 4B ?

DRAgon734465502, 2019-12-13:
Can I use mk_arcade_joystick_rpi at raspberry pi 4B ?

DRAgon734465502, 2020-01-02:
QAQ?

dcoueron, 2020-01-24:
Can't make it work either. Did you manage to ?

DRAgon734465502, 2020-02-14:
still can not

Gemba, 2020-03-19:
For those who stranded here: This [fork](https://github.com/cmitu/mk_arcade_joystick_rpi/tree/retropie) by cmitu is much better maintained. Use either the branch retropie or retropie-hotkeybtn. 

Cmitu's fork also incorporates the pending PR #72 and #74 from here for recent kernels.

DRAgon734465502, 2020-03-24:
sorry , I can not understand what do you mean /

You mean we still need  wait  the update ?

DRAgon734465502, 2020-05-08:
the retropie official web have update the newest image for 4b
https://retropie.org.uk/download/
But mk_arcade_joystick_rpi still can not work at  the newest 4.6 4B  image

Gemba, 2020-05-09:
The support for RPi4 is in the fork I mentioned earlier, since two days.
Better turn to the [retropie forums](https://retropie.org.uk/forum/) for a detailed discussion if it still does not work.

ian57, 2020-05-10:
This github is deprecated. Recalbox has switched to gitlab for a mong time. the 0.18 release of mkarcade is already ok for kernel > 4.15 https://gitlab.com/recalbox/mk_arcade_joystick_rpi/-/tree/v0.1.8

DRAgon734465502, 2020-07-01:
> This github is deprecated. Recalbox has switched to gitlab for a mong time. the 0.18 release of mkarcade is already ok for kernel > 4.15 https://gitlab.com/recalbox/mk_arcade_joystick_rpi/-/tree/v0.1.8
I have tested 4.15 ， it do not work at 4B