recalbox/mk_arcade_joystick_rpi #80: mk_arcade_joystick_rpi not work at retropie 4.6

DRAgon734465502, 2020-07-25:
I have tried  mk_arcade_joystick_rpi  at  retropie 4.6

but the driver do not work， 

Who know why ？

KGOrphanides, 2021-04-15:
I assume you're having this issue with the version installed from RetroPie's package managers. If that's the case, and if you're using a Raspberry Pi 4 or 400, the way in which the GPIO handles pull-up switches has changed.

Stick the following in a file called /home/pi/pullup.sh 

`#!/bin/bash
# fix for mk_arcade_joystick_rpi driver issue inherited from db9_gpio_rpi see https://github.com/marqs85/db9_gpio_rpi/issues/8
# RPi 4 and 400 need this tweak to speak to mk_arcade_joystick_rpi driver gpio connected arcade sticks and buttons as some inputs need an explicit pullup
# ensure that you're applying the pullups to the correct pins - this is for a standard deployment of the driver with two sets of buttons 
#
# Use:
# chmod pullup.sh +x
# add /path/to/pullup.sh to /etc/rc.local to load on boot

# Player 1 Joyport /dev/input/js0
raspi-gpio set 2 ip pu  #hotkey
raspi-gpio set 4 ip pu
raspi-gpio set 17 ip pu
raspi-gpio set 27 ip pu
raspi-gpio set 22 ip pu
raspi-gpio set 10 ip pu
raspi-gpio set 9 ip pu
raspi-gpio set 25 ip pu
raspi-gpio set 24 ip pu
raspi-gpio set 23 ip pu
raspi-gpio set 18 ip pu
raspi-gpio set 15 ip pu
raspi-gpio set 14 ip pu

# Player 2 Joyport /dev/input/js1
raspi-gpio set 3 ip pu  #hotkey
raspi-gpio set 11 ip pu
raspi-gpio set 5 ip pu
raspi-gpio set 6 ip pu
raspi-gpio set 13 ip pu
raspi-gpio set 19 ip pu
raspi-gpio set 26 ip pu
raspi-gpio set 21 ip pu
raspi-gpio set 20 ip pu
raspi-gpio set 16 ip pu
raspi-gpio set 12 ip pu
raspi-gpio set 7 ip pu
raspi-gpio set 8 ip pu`

chmod /home/pi/pullup.sh +x
nano /etc/rc.local

Above the exit line, add:
`/home/pi/pullup.sh`

Reboot.

museany, 2021-08-15:
no body know why ,you should look at libaray by pi 4B